SlimeVR/SlimeVR-Tracker-ESP #6: Implement proper serial commands handling

Eirenliel, 2021-06-12:
Need to handle a few commands from serial port for debug, testing and configuration. Need to implement the system as a whole, and a few necessary commands:

* Health report
* Reboot
* Factory reset
* Set up WiFi credentials

adigyran, 2021-07-21:
can you add current tracker IP command also?

Kamilake, 2022-03-15:
Reboot, Factory reset seems to have been completed in #101 #102.

Bluscream, 2022-04-15:
A "GET CONFIG" command would be nice as well

Eirenliel, 2022-04-15:
True.

loucass003, 2022-07-10:
What is needed in the health statuts ?
what infos do we want to put in it

Eirenliel, 2022-07-10:
Which server it is connected to, status of IMUs, at the very least.

loucass003, 2022-07-10:
Im gonna work on the last one this week, while i am waiting for new gui stuff

Eirenliel, 2022-12-28:
GET HEALTH should be GET TEST with some additional info, with all IMUs and server connection status.

PR #217 (2023-01-06): Multiple debugging updates to serial interface
Fixes #6 and #215

Adds some minor fixes to logging for testing and debugging purposes.
closed 2023-01-06 by Eirenliel