SlimeVR/SlimeVR-Tracker-ESP #12: EEPROM handling causes a LoadStore Error

ThePheonixGuy, 2021-08-07:
Hello! Trying to build a DIY version of this but something is not very happy. 

When running the project on an esp32, I'm getting the following error: (backtrace debugging turned on for help)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0a62  PS      : 0x00060530  A0      : 0x800d0bcc  A1      : 0x3ffb1f60
A2      : 0x3ffc1730  A3      : 0x000186a0  A4      : 0x0000000d  A5      : 0x00000000  
A6      : 0x00000003  A7      : 0x00000000  A8      : 0x3f400120  A9      : 0x3ffb84d5
A10     : 0x000000ff  A11     : 0x000000ff  A12     : 0x00000000  A13     : 0x00000064  
A14     : 0x00000064  A15     : 0x60013000  SAR     : 0x0000001c  EXCCAUSE: 0x00000003
EXCVADDR: 0x3f400120  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d0a62:0x3ffb1f60 0x400d0bc9:0x3ffb1f80 0x400d713a:0x3ffb1fb0 0x4008a30e:0x3ffb1fd0
  #0  0x400d0a62:0x3ffb1f60 in DeviceConfig* const& EEPROMClass::get<DeviceConfig* const>(int, DeviceConfig* const&) at src/configuration.cpp:58
      (inlined by) getConfigPtr() at src/configuration.cpp:44
  #1  0x400d0bc9:0x3ffb1f80 in setup() at src/main.cpp:99
  #2  0x400d713a:0x3ffb1fb0 in loopTask(void*) at C:\Users\User-PC\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:18
  #3  0x4008a30e:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Any assistance? Tested on an ESP DOIT DevKit v1 as well as a Firebeetle ESP32 V4, both thow the same issue.

Eirenliel, 2021-08-07:
@ThePheonixGuy i broke some stuff there when refactoring config, and honestly not sure how to fix. Last commit might work, i think?

ThePheonixGuy, 2021-08-07:
I actually have found a fix. Will update this issue with a PR shortly. Will try the last version too to see if it helps.

ThePheonixGuy, 2021-08-07:
Created the PR.

ThePheonixGuy, 2021-08-08:
Closed with #13

closed 2021-08-08 by ThePheonixGuy