SlimeVR/SlimeVR-Tracker-ESP #14: Issue with Sensor data from MPU-9250

ThePheonixGuy, 2021-08-08:
Hello Again!

I've been trying to set up some trackers of my own with this. 
I have gotten to a point where its mostly stable for one sensor, for a while (+- 5 minutes). However, all other sensors after a short time (within a few seconds) start reporting "NaN NaN NaN" for their rotation value. 

I've run a short sketch on my esp's to ensure the MPU's are wired and configured correctly - and they are, showing data across the range. 

I added some debug print lines to the MPU9250Sensor::getMPUScaled() call to show me the raw data, and then another to show me what gets sent over the wire. 

This results in values like such:

```[DEBUG] Raw data: 32767,32767,-3520A, -221,-256,-1370G, -129,-129,16626M
[DEBUG] Sent data: -0.00,39145.87,32490.37RM, 0.00,0.00,1.00A, -0.00,0.77,0.64M, nan,nan,nanQ.
[DEBUG] Raw data: 32767,32767,-3476A, -308,-243,-1429G, -129,-129,27890M
[DEBUG] Sent data: -0.00,61243.93,54674.87RM, 0.00,0.00,1.00A, -0.00,0.75,0.67M, nan,nan,nanQ.
[DEBUG] Raw data: 32767,32767,-3532A, -178,-211,-1414G, -129,-129,13554M
[DEBUG] Sent data: -0.00,33119.12,26440.05RM, 0.00,0.00,1.00A, -0.00,0.78,0.62M, nan,nan,nanQ.
[DEBUG] Raw data: 32767,32767,-3524A, -130,-236,-1419G, -129,-129,15602M
[DEBUG] Sent data: -0.00,37136.96,30473.60RM, 0.00,0.00,1.00A, -0.00,0.77,0.63M, nan,nan,nanQ.
[DEBUG] Raw data: 32767,32767,-3600A, -296,-218,-1411G, -129,-129,-3855M
[DEBUG] Sent data: 0.00,-1034.39,-7847.05RM, 0.00,0.00,1.00A, 0.00,-0.13,-0.99M, nan,nan,nanQ.
[DEBUG] Raw data: 32767,32767,-3688A, -339,-224,-1399G, -129,-129,-26383M
[DEBUG] Sent data: 0.00,-45230.51,-52216.04RM, 0.00,0.00,1.00A, 0.00,-0.65,-0.76M, nan,nan,nanQ.
[DEBUG] Raw data: 32767,32767,-3604A, -150,-230,-1350G, -129,-129,-4879M
[DEBUG] Sent data: 0.00,-3043.31,-9863.82RM, 0.00,0.00,1.00A, 0.00,-0.29,-0.96M, nan,nan,nanQ.
[DEBUG] Raw data: 32767,32767,-3636A, -195,-219,-1436G, -129,-129,-13071M
[DEBUG] Sent data: 0.00,-19114.62,-25998.00RM, 0.00,0.00,1.00A, 0.00,-0.59,-0.81M, nan,nan,nanQ.
[DEBUG] Raw data: 32767,32767,-3624A, -335,-229,-1390G, -129,-129,-9999M
[DEBUG] Sent data: 0.00,-13087.88,-19947.68RM, 0.00,0.00,1.00A, 0.00,-0.55,-0.84M, nan,nan,nanQ.```

Any ideas? Again - checked the sensor with another sketch to ensure it is working and all are working perfectly fine.

Eirenliel, 2021-08-08:
It's not calibrated and there is a division by 0 with default calibration matrix (all 0). This is something I will fix soon by adding server-side calibration, but I didn't have time to work on this yet.

adigyran, 2021-08-08:
You have to run in 6500 mode with fork https://github.com/masterex1000/SlimeVR-Tracker-ESP

ThePheonixGuy, 2021-08-08:
Does that sacrifice the magnetometer though? Thanks for the link!

Blixtdraken, 2021-08-09:
The use of magnetometers on MPU 9250 is an ongoing project. At some point it will be utilized.

Eirenliel, 2021-08-16:
This is required to utilize MPU-9250's magentometer: https://github.com/SlimeVR/SlimeVR-Server/issues/13

Eirenliel, 2022-04-02:
Have been implemented awhile ago in many isses.

closed 2022-04-02 by Eirenliel