SlimeVR/SlimeVR-Tracker-ESP #34: Flash failure from Wifi writing credentials

sctanf, 2021-11-04:
Wifi.begin() writes credentials to flash every time it is called, on some boards this causes the flash to fail quickly, resulting in Wifi.begin() resetting the board.

Calling Wifi.persistent(false) before Wifi.begin() (and potentially other wifi functions) stops the credentials from being written. On boards with failing flash, this will continue working.

deiteris, 2021-11-04:
Related issue:
https://github.com/esp8266/Arduino/issues/1054

Kamilake, 2021-11-06:
That's interesting. It doesn't seem to happen on ESP32.

Does that cause the whole board to boot fail? Or does it only corrupt credentials?
Do I(that board) need to Wifi.persistent(false) on every run except when performing a write operation?

deiteris, 2021-11-06:
Similar issue seems to be occurred on Wemos D1 Mini when powered via micro-USB port, but works ok on battery. Though `Wifi.persistent(false)` didn't help.
https://discord.com/channels/817184208525983775/823930029070876736/905833475741982721

sctanf, 2021-11-06:
@Kamilake No, the board resets if you try to access damaged flash, so credentials can no longer be written. Not sure if it can still be read. Otherwise it appears to work fine

Eirenliel, 2021-11-15:
Need to review Wifi.begin() code for different frameworks and see how to start wifi properly without it being re-written, but still being able to read it on begin and write on credentials change.

Vyolex, 2021-11-16:
On the `espressif8266` platform this should not be the case. Only when SSID and the passphrase are provided will Wifi.begin() write to flash if persistence is not set to false. When not providing these parameters it should only read from flash (code below).

```cpp
wl_status_t ESP8266WiFiSTAClass::begin() {

    if(!WiFi.enableSTA(true)) {
        // enable STA failed
        return WL_CONNECT_FAILED;
    }

    ETS_UART_INTR_DISABLE();
    wifi_station_connect();
    ETS_UART_INTR_ENABLE();

    if(!_useStaticIp) {
        wifi_station_dhcpc_start();
    }
    return status();
}```

Vyolex, 2023-06-16:
Can this be closed? Neither esp32 nor esp8266 does any writing to memory when wifi.begin() is called without arguments. (In fact the function does not even reference if persistence is set to true or false in either function.)

Eirenliel, 2023-06-16:
Did we used to use WiFi.begin(SSID, pass)? Or was this open on error, @Vyolex?

Vyolex, 2023-06-17:
IDK if it used to be different but atm we only call WiFi.begin(SSID, pass) in `setWiFiCredentials` and in `upkeep` (the latter only when the ssid and pass are defined in code). So default behaviour should not cause data to be written to flash every single boot, only on sending credentials from server.

closed 2023-06-17 by Eirenliel