SlimeVR/SlimeVR-Tracker-ESP #44: Add wifi signal strength indicator

deiteris, 2021-11-30:
It would be useful to see signal strength to better diagnose connectivity issues. Arduino framework provides a method to get signal strength of the current connection: https://www.arduino.cc/en/Reference/WiFiRSSI

Kamilake, 2021-12-01:
Yes! it certainly works as necessary and expected.
I'm thinking about what to do to make it easier for users to see intuitively)

I am currently testing by inserting the following code into the clientUpdate function.
```cpp
if (WiFi.status() != WL_CONNECTED) {
    Serial.println("DISCONNECTED");
}
else {
int rssi = WiFi.RSSI();
    Serial.print("Signal Strenth: ");
    Serial.print(100+rssi);
    Serial.print("% (");
    Serial.print(rssi);
    Serial.println(")");
}
```
(RSSI value is not linear so I think there is a limit to displaying it directly as %)

TheButlah, 2021-12-01:
Is it better to display signal strength, or instead just latency + dropped packet rate?

Kamilake, 2021-12-01:
> Is it better to display signal strength, or instead just latency + dropped packet rate?

I'm curious about that too
For experts with DIY, it will be easier to understand and more useful to show dropped+rssi
But for beginners or ordinary people who don't, signal strength% + latency will be friendly.

TheButlah, 2021-12-01:
% of packets dropped can be mapped to a wifi signal indicator just as well as a wifi signal strength can

Eirenliel, 2021-12-01:
we can't measure dropped packets, since they're not tracked currently. we can start tracking them, but also eh... latency is displayed already, that's PING. signal strength can be added, ye, i'm up for this

Eirenliel, 2022-01-27:
Implemented.

closed 2022-01-27 by Eirenliel