SlimeVR/SlimeVR-Tracker-ESP #69: Make sensor fusion algorithm IMU-independent

TheButlah, 2022-01-05:
There is a sensor fusion algorithm being used right now to fuse the dmp-enabled 6dof gyro + accel data in the MPU9250 with the magnetometer.

Would it be possible to extract this code into an IMU-agnostic class? That way it could be easier used in other IMUs. Also, my understanding was that mahoney can be used for 9dof or for 6dof. Can we make the class able to do either? This would be useful because the ICM-20602 does not have a DMP so it must run the fusion on the ESP.

Eirenliel, 2022-01-05:
Isn't it already done in #55 and #68?

TheButlah, 2022-01-05:
Thanks for drawing my attention to those! Once merged, they would definitely close this issue :)
Still navigating my way around the codebase so I'm sticking to the `main` branch for now

Boronare, 2022-01-05:
As I know, Mahony filter fusion doesn't use DMP. It only uses accel/gyro/mag values.
What I added in #55 is using DMP. You can also use Mahony method instead of DMP fusion with define statement.

deiteris, 2022-02-02:
I think this can be closed. We have merged the PRs that add Mahony and Madgwick filters as a separate library, so they can be used by everyone.

Eirenliel, 2022-02-02:
True.

closed 2022-02-02 by Eirenliel