SlimeVR/SlimeVR-Tracker-ESP #71: Fancy wifi provisioning

Eirenliel, 2022-01-13:
This is both for firmware and software. Here are features that needs to be implemented:

- [ ] Tracker creates an AP with configuration protal, if it can't connect to any wifi network (pr #59 already implements this)
- [ ] Tracker can be commanded to provision nearby trackers. With this command it tries to connect to all unconfigured trackers (it gudges by an AP name) and sends them credentials of its own wifi network
  - [ ] Server should be able to command trackers to provision other trackers remotely or via USB
- [ ] When tracker is connected via USB, but it doesn't have a wifi connection, a popup offering wifi configuration should pop up
  - [ ] This pop up should let user select out of found wifi networks, it should have an checkbox "Connect other trackers", see above

Eirenliel, 2022-04-11:
[image] [Fancy Wi-Fi Provisioning](https://trello.com/c/Vc2GPIcA/12-fancy-wi-fi-provisioning)

loucass003, 2022-07-10:
``` When tracker is connected via USB, but it doesn't have a wifi connection, a popup offering wifi configuration should pop up```

Is this still a thing with the wifi provisioning in the gui. Because those two approches would get in conflict. 

if you are provisioning new trackers you dont want the wifi popup to show up on the wifi page of the new gui.

Eirenliel, 2022-07-10:
Yeah, that shouldn't be on the wifi page, that should be when you're not in the wizard.

electromuis, 2024-05-14:
Just added some support for wifi scanning in the gui. Might be relevant here: https://github.com/SlimeVR/SlimeVR-Server/pull/1033