SlimeVR/SlimeVR-Tracker-ESP #93: OTA Push does not update WiFi Settings

WindDrake, 2022-02-06:
Been banging on this for an hour or so now.

If I push an update over OTA, the tracker will take the update, but it will not change the hardcoded WiFi settings on the tracker if I've changed them in platformio.ini.

Dismantling my tracker and looking at the serial port monitor, the tracker was looking for the old WiFi AP, not the new one.

Pushing the same firmware over USB does not fix this either. The only way for me to fix it was to use the WiFi button in Server. Had to dismantle all 6 of my trackers.

Eirenliel, 2022-02-06:
Hardcoded wifi credentials are a "fallback" of sorts. The tracker will try to connect to the last wifi it was connected to, and then if it can't - to hardcoded one. There aren't any betteer way to do this, one of them should be a priority. This is only a bug if it doesn't connect to hardcoded wifi even if the last one couldn't connect.

WindDrake, 2022-02-06:
Maybe a flag to uncomment that forces a re-read of the hardcoded credentials next boot?

Eirenliel, 2022-02-06:
This can be done, okay.

unlogisch04, 2022-07-10:
In general, the WiFi is saved in the persistent storage of the ESP. If it is able to connect from that, the connection is successful.
We should address that in https://github.com/SlimeVR/SlimeVR-Tracker-ESP/issues/128

unlogisch04, 2023-01-23:
Affects only hardcoded WiFi settings. 
Workaround is, to go with the tracker out of the old WiFi Range. Then it try to reconnect with the new WiFi Settings from the hardcoded credentials.