SlimeVR/SlimeVR-Tracker-ESP #127: Unable to use 9250s as auxiliary  if the main imu is 20948

fullxu, 2022-03-14:
[DEBUG] [ICM20948Sensor:0] DMP initialized
[DEBUG] [ICM20948Sensor:0] Using 9 axis configuration
[DEBUG] [ICM20948Sensor:0] Enabled DMP sensor for sensor orientation
[DEBUG] [ICM20948Sensor:0] Set Quat9 to 100Hz frequency
[DEBUG] [ICM20948Sensor:0] FIFO Enabled
[DEBUG] [ICM20948Sensor:0] DMP Enabled
[DEBUG] [ICM20948Sensor:0] Reset DMP
[DEBUG] [ICM20948Sensor:0] Reset FIFO
[INFO ] [MPU9250Sensor:1] Connected to MPU9250 (0x71) at address 0x69
[INFO ] [MPU9250Sensor:1] Flip front to confirm start calibration
[DEBUG] [MPU9250Sensor:1] Starting calibration...
[DEBUG] [MPU9250Sensor:1] Gathering raw data for device calibration...
[INFO ] [MPU9250Sensor:1] Put down the device and wait for baseline gyro reading calibration
[INFO ] [MPU9250Sensor:1] Gently rotate the device while it's gathering accelerometer and magnetometer data
[DEBUG] [MPU9250Sensor:1] Calculating calibration data...

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

last failed alloc call: 4020D112(24000)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x4020d112

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0006f7e0
~ld

TheDevMinerTV, 2022-03-14:
Seems like we're running out of memory here :/

TinyOra, 2022-03-14:
> weren't memory optimized for 9250 already?

Yep , it can be used alone , but is my main IMU is 20948, and 9250 as aux, i can't finish calibration progress

TinyOra, 2022-03-14:
I made a separate tracker for the waist, But the feet are difficult :<

LETS-BEE, 2022-03-28:
+ 9250 main, 6500 aux too. can't finish calibration process.

Eirenliel, 2022-03-28:
So we need to reduce memory usage of Magneto...

deiteris, 2022-03-28:
Magneto library doesn't consume much memory. The problem is most likely that DMP image is allocated on stack and takes a lot of space. We should try allocating it on heap and freeing when not needed.

unlogisch04, 2022-03-31:
With changing to pgmspace i could free about 14'304 byte heap on ESP8266 no change on ESP32 i think this was already fine before. #138

TinyOra, 2022-04-02:
> Should be fixed. Reopen if the issue persists with the latest main.

After this fixed, new issue produced :
When finished calibration, it will work fine with 20948, but if you restart tracker, 9250 will not work anymore(just like you never do a calibration process)
Now the 9250 need recalibrate every time restart the tracker, otherwise it will not work
-------------------------------------------------------------
[OK] Port opened
[DEBUG] [ICM20948Sensor:0] DMP initialized
[DEBUG] [ICM20948Sensor:0] Using 9 axis configuration
[DEBUG] [ICM20948Sensor:0] Enabled DMP sensor for sensor orientation
[DEBUG] [ICM20948Sensor:0] Set Quat9 to 100Hz frequency
[DEBUG] [ICM20948Sensor:0] FIFO Enabled
[DEBUG] [ICM20948Sensor:0] DMP Enabled
[DEBUG] [ICM20948Sensor:0] Reset DMP
[DEBUG] [ICM20948Sensor:0] Reset FIFO
[INFO ] [MPU9250Sensor:1] Connected to MPU9250 (0x71) at address 0x69
[DEBUG] [MPU9250Sensor:1] Enabling DMP...
[DEBUG] [MPU9250Sensor:1] DMP ready! Waiting for first interrupt...
[INFO ] [WiFiHandler] Setting up WiFi
[INFO ] [WiFiHandler] Loaded credentials for SSID potato and pass length 8
[DEBUG] [WiFiHandler] Status: 7
[NOTICE] OTA updates allowed
.[ERROR] [ICM20948Sensor:0] Sensor timeout I2C Address 0x68
...[INFO ] [WiFiHandler] Connected successfully to SSID 'potato', ip address 192.168.1.30
----------------------------------------------------------------------------------------
Logs seems to be normal except i got a "Sensor timeout", but in fact, the 20948 is working fine 
I had tried it on the other 20948 Main trackers,  same issue

TinyOra, 2022-04-02:
> Should be fixed. Reopen if the issue persists with the latest main

I had done some more tests,  I noticed it has some probability that it will work, not completely out of work
The details is in discord
https://discord.com/channels/817184208525983775/912982617010352159/959802967928754176

Sorry to bother you so much, thanks for a lot

unlogisch04, 2022-04-02:
seems a different issue with the configuration storage (eeprom). PR #134 should fix that anyway.

Eirenliel, 2022-04-02:
#134 in main, see if this works

TinyOra, 2022-04-02:
> #134 in main, see if this works

It's work ,thanks

closed 2022-04-02 by Eirenliel