SlimeVR/SlimeVR-Tracker-ESP #146: Add ID command

Eirenliel, 2022-04-04:
An ID command sent from the server should force tracker to light up the LED for identification. Command can contain trackerId but the tracker can ignore it if there are no way to identify only extension. Command should be acknowledged by replying it back to the server. Another command with state off should stop the process and again be acknowledged to the server.

Eirenliel, 2022-04-11:
[image] [Add ID command to tracker](https://trello.com/c/av0PCiCA/14-add-id-command-to-tracker)

TheDevMinerTV, 2022-04-24:
This should be added into the new protocol tbh, there's no need to change the current netcode just to add this and then reimplement it later.

gorbit99, 2025-05-05:
Is this still relevant?

Eirenliel, 2025-05-05:
Yes, and it will be also interesting for smol slimes

gorbit99, 2025-05-05:
@Eirenliel does it make sense for the packet to contain trackerId? Not like the extension would have a led to light up. Instead what about just a boolean for "light up led" vs "dim led"?

Eirenliel, 2025-05-05:
Eh it's better to have the tracker id and let the tracker decide what to do with it. We support trackers with many extensions and many designs.

sctanf, 2025-05-05:
if support is needed for smols, an issue can be created there

Eirenliel, 2025-05-06:
@sctanf yeah but for now we can have it in the protocol on both sides and see how we want to actually use it.