SlimeVR/SlimeVR-Tracker-ESP #148: Idle Drift compensation

beareogaming, 2022-04-10:
Would it be possible to implement some sort of idle compensation for drift, my thought process was that if you stay at, lets say 100 degree of rotation, and it stays between 100±1 for 5 minutes that the tracker would go into an "idle" where if after more than 5 minutes of being in that range, it shifts outside it by a single degree for more than a minute over a period of time, that it would deem it as drift and put a 1 degree offset on the tracker in the opposing direction.

of course in order to prevent accidental movement, if you shift more than 1 degree outside the range it would disable the idle and would require that tracker to stay still to the ±1 degree for more than 5 minutes to go back into it, this could help for people sitting/laying for extended times without a need to reset as often.

from my short experience the only time i have drift is over a period of time ~5m minimum per degree of rotation in the worst case, 

the only way this should cause issues is if someone continually moves one degree at a time in a direction, perhapse make a setting to toggle on when you are sitting still for periods of time/adjust the 5 minute window

Eirenliel, 2022-04-11:
The main problem is that 99% of drift happens during movement, not during idle. We can detect idle using accelerometer, that's how its done on BNO085, afaik, and it's more reliable. But wht we can't do is make gyroscope less noisy...

deiteris, 2022-04-11:
There's an idea to recalibrate gyroscope when idling (using accelerometer), so it's less affected by environmental factors. I had partial success, but my development is a bit suspended rn.