SlimeVR/SlimeVR-Tracker-ESP #160: Indicate when IMU is calibrated

Louka3000, 2022-05-24:
Send a packet once the IMU is calibrated (currently lies about it) so that the server can tell the user when their tracker is calibrated in GUI **or** don’t send rotation packets until IMU is calibrated. Maybe also do a certain blink pattern once calibrated.

unlogisch04, 2023-01-23:
Is this still valid?

Louka3000, 2023-01-23:
https://github.com/SlimeVR/SlimeVR-Tracker-ESP/pull/222 would be about it/replace it I believe

Kamilake, 2023-01-24:
> #222 would be about it/replace it I believe

MPU series uses a different calibration method from BNO and requires calibration every time it starts, isn't it different?

Louka3000, 2023-01-24:
BNOs also need calibration everytime it starts as it currently is

Eirenliel, 2023-01-24:
> BNOs also need calibration everytime it starts as it currently is

We have conflicting info if it does. I should maybe talk to CEVA.

Louka3000, 2023-01-24:
Hmm well that’s what I’ve been told for 2 years by people on Discord :p

Eirenliel, 2023-01-24:
Yeah, and CEVA also confirms it, see my reply in #222

gorbit99, 2025-05-05:
Should be solved in #384