SlimeVR/SlimeVR-Tracker-ESP #161: Allow tracker rotation to be changed at runtime

wolfiestyle, 2022-06-03:
Currently there is a hardcoded "IMU_ROTATION" parameter that needs to be set at compile time. I don't see why this parameter shouldn't be a setting on the server instead. It would be really useful for prototyping, or if you accidentally put on your tracker on the wrong side.

ButterscotchV, 2022-06-03:
This has been discussed before, imo it should be hard set in firmware because it shouldn't change. Once you build it and have it wired, it shouldn't change frequently. Maybe there's a chance for it in the future? But I don't quite see a use for it for most people.

Loler920a, 2022-06-03:
I'm hoping this would be resolved when "dynamic mounting/orientation" makes an improvement and return. I'm hopeful it would be 3 axis. But I think the maths requires 1 axis to be known/locked to calculate this.

But going back to the issue at hand. 
For locking it on firmware, I'm on the fence about this. 

But it could be better down the line in case anyone switches a main unit upside down. But then why not just take it off and flip it around?

For people buying trackers from another party maybe? But it's usually best to reflash it anyways as the firmware is outdated, or hardcoded or not configured properly. Heck no firmware at all sometimes.

Though I still see the convenience of just quickly changing things in server. As well as possibly reducing the amount of space being taken up in the Arduino.

Eirenliel, 2022-06-03:
This should be possible to figure out during reset, because accelerometer always knows where the down is anyway, just no one implemented it yet.

unlogisch04, 2024-06-19:
Mounting Reset takes care of it.

unlogisch04, 2024-06-19:
Feature Automatic Mounting Reset in the Server takes care of it.

closed 2024-06-19 by unlogisch04