SlimeVR/SlimeVR-Tracker-ESP #187: Unable to save AUX IMU calibration data when using MPU9250*2

Disappear9, 2022-08-10:
### Setup: 
ESP32 MPU9250*2
### Version: 
main
### Tried:
clear flash and reflash firmware.  (x)
swap esp32  (x)
### Screenshots:
calibration data saved
[image]

after restart, (No calibration data found for sensor 1):
[image]

unlogisch04, 2022-08-10:
Thank you for your great report. With the PR #188 it should work now.

Disappear9, 2022-08-11:
Tested, it's working now.

````
[INFO ] [SlimeVR] SlimeVR v0.2.2 starting up...
[INFO ] [Configuration] Found up-to-date configuration v1
[DEBUG] [Configuration] Found calibration data directory
[DEBUG] [Configuration] Found sensor calibration for MPU9250 at index 0
[DEBUG] [Configuration] Found sensor calibration for MPU9250 at index 1
[INFO ] [Configuration] Loaded configuration
[INFO ] [MPU9250Sensor:0] Connected to MPU9250 (reported device ID 0x71) at address 0x68
[DEBUG] [MPU9250Sensor:0] Enabling DMP...
[DEBUG] [MPU9250Sensor:0] DMP ready! Waiting for first interrupt...
[INFO ] [MPU9250Sensor:1] Connected to MPU9250 (reported device ID 0x71) at address 0x69
[DEBUG] [MPU9250Sensor:1] Enabling DMP...
[DEBUG] [MPU9250Sensor:1] DMP ready! Waiting for first interrupt...
[INFO ] [WiFiHandler] Setting up WiFi
[INFO ] [WiFiHandler] Loaded credentials for SSID  and pass length 63
[DEBUG] [WiFiHandler] Status: 6
[NOTICE] OTA updates allowed
..[INFO ] [WiFiHandler] Connected successfully to SSID ' ', ip address 
[INFO ] [UDPClient] Looking for the server...
````

okay? why github just auto closed this issue for me😂

PR #188 (2022-08-11): Unable to save AUX IMU calibration data when using MPU9250*2
Fixes #187
commit d2d40f46 (2022-08-11): Unable to save AUX IMU calibration data when using MPU9250*2 (#188)

Fixes #187