SlimeVR/SlimeVR-Tracker-ESP #198: Body now rotates around the head when you turn left/right.

ThreadOfFate, 2022-10-02:
Issue seems to be introduced between these commits.
https://github.com/SlimeVR/SlimeVR-Tracker-ESP/commit/9d93df6e6a0f3c0f68466669a100667925b4d801
https://github.com/SlimeVR/SlimeVR-Tracker-ESP/commit/b8bd02e60d01312634fa71b12512b7556cd0dedf

Both reports of this issue are from ICM 20948 users.

[image]
[image]

TheDevMinerTV, 2022-10-02:
These are the changes between those commits: https://github.com/SlimeVR/SlimeVR-Tracker-ESP/compare/9d93df6e6a0f3c0f68466669a100667925b4d801..b8bd02e60d01312634fa71b12512b7556cd0dedf

unlogisch04, 2022-10-02:
I do expect this line to be the reason for the change. https://github.com/SlimeVR/SlimeVR-Tracker-ESP/compare/b8bd02e60d01312634fa71b12512b7556cd0dedf..9d93df6e6a0f3c0f68466669a100667925b4d801#diff-d057e2d1cf9af1b161e90eb4c9f63baa4658f80e6351f90a862c5a14dadd4622R410

unlogisch04, 2022-10-05:
I work on this https://github.com/unlogisch04/SlimeVR-Tracker-ESP/tree/ICM20948_MagnetometerFix

closed 2022-10-27 by Eirenliel