SlimeVR/SlimeVR-Tracker-ESP #203: Trackers not connecting without WiFi reboot

ButterscotchV, 2022-10-06:
Discussion is on Discord: https://discord.com/channels/817184208525983775/1027559999574114376

There seems to be an issue with specific routers where a reboot is required to get Slime trackers to connect.

Routers known to have this issue:
- TP-Link Archer AXE75 (AXE5400)
- TP-Link Archer C4000 (AC4000)

Routers that require to be set to "Legacy" Wireless Mode:
- ASUS TUF-AX5400
- ASUS RT-AX92U

Eirenliel, 2022-10-07:
seem to be related: https://github.com/esp8266/Arduino/issues/8412

ButterscotchV, 2022-10-08:
Stumbled upon a possible solution from the esp8266 thread, apparently these issues are fixed in the latest esp8266 Arduino version and you can switch to that in PlatformIO by changing the board config in `platformio.ini`

From
```ini
[env:esp12e]
platform = espressif8266
board = esp12e
; Comment out this line below if you have any trouble uploading the firmware
; and if it has a CP2102 on it (a square chip next to the usb port): change to 3000000 (3 million) for even faster upload speed
upload_speed = 921600
```
to
```ini
[env:esp12e]
platform = espressif8266
board = esp12e
; Comment out this line below if you have any trouble uploading the firmware
; and if it has a CP2102 on it (a square chip next to the usb port): change to 3000000 (3 million) for even faster upload speed
upload_speed = 921600
platform_packages =
  platformio/framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
```

ButterscotchV, 2022-11-16:
Should be fixed with release 0.3.0

closed 2022-11-16 by ButterscotchV

ButterscotchV, 2022-11-17:
Re-opening at the request of @nullstalgia, the source of the problem still isn't quite found

TheGuardianWolf, 2023-01-14:
[image]

I am able to connect without the fallback by switching to 20MHz simply. Perhaps this chip is not compatible with 40MHz? ESP32 is fine though.

Before on 20/40MHz
[image]

After:
[image]