SlimeVR/SlimeVR-Tracker-ESP #210: Issues when cloning repo

leonbeer, 2022-11-11:
Hey there! 

I've been getting random include errors when trying to build. 
There's always a random file missing, according to PlatformIO.
Cloning the report again does not fix the issue, then there is another file missing

Kamilake, 2022-11-11:
May I ask how you copied it?
(ex. "Download ZIP" Button, `gh repo clone`,  `git clone` or some GUI...)

TheDevMinerTV, 2022-11-11:
> Hey there!
> 
> I've been getting random include errors when trying to build. There's always a random file missing, according to PlatformIO. Cloning the repo again does not fix the issue, then there is another file missing

Please tell us what the error is, otherwise we can't really help you.

leonbeer, 2022-11-11:
It's always a different error. 
The most recent error was that
It couldn't find pins_arduino.h
(Missing File)

leonbeer, 2022-11-11:
I also used the GIT integration of VSCode to clone the repository

TheDevMinerTV, 2022-11-11:
This `could not find pins_arduino.h` error indicates that you did not properly set up your PlatformIO environment. [Please follow our documentation.](https://docs.slimevr.dev/firmware/setup-and-install.html)

Also, please copy-paste errors the next time instead of giving us a rough error message, we can help you more with those.

leonbeer, 2022-11-11:
The Environment is setup exactly as shown in the documentation. 

I am sorry for the lack of the exact error message,

leonbeer, 2022-11-11:
Issue resolved after deleting the .PlatformIO folder and reinstalling.

closed 2022-11-11 by leonbeer