SlimeVR/SlimeVR-Tracker-ESP #211: BNO08X Calibration

TDW2401, 2022-11-27:
I've noticed while spinning through BNO08X datasheets that it recommends a self calibration from the user 
*For best motion tracking performance, it is recommended to calibrate the BNO08X. Since each MEMS sensor
part has different individual characteristics, each device using the BNO08X must be calibrated individually.*
This seems to be a one time thing.
Ceva suggests that even if this isn't done, the BNO will still work using factory calibration but I believe it could be worth a look in to if a user calibration could improve overall drift time.
I would be interested to hear others opinions though.

Eirenliel, 2022-11-27:
We also need to look into disabling auto-calibration some time after boot, since it may produce some unwanted results like broken accelerometer when stomping your leg.

nullstalgia, 2022-11-27:
I have attempted to do calibration with my BNO085s due to experiencing unusual drift times (sometimes ~10-15 mins when just sitting down and talking).

What I would recommend based on my experience:

When you click on a tracker in the Home, there could be a button to "Start Calibration" that opens a modal that shows the BNO-reported "Accuracy" for Gyro, Mag, and Accel, and a "Save" button for when they're as high as it will go. (it goes from 0 to 3, 0 being unreliable, 3 being high reliability).

Animated renders or a demo recording of the calibration movements described in the datasheet or CEVA procedure doc 1000-4044 would be best for users.

In any case, as long as we don't tell the BNO to save the new calibration unless the user confirms it on the UI, we should be able to avoid most adventurous users accidentally goofing their trackers' calibration.