SlimeVR/SlimeVR-Tracker-ESP #218: MPU9250 does not work properly after this commit

TinyOra, 2023-01-18:
Problem commits: fd3e463a9c60411715ed3ff2b333158ada86c225

After flashing ,The 9250 's yaw data is very confusing, If I rotate to the left, the data increases first and then decreases after a while,  and tends to return to its original value. If I rotate to the right, just the opposite

In actual use, the tracker rotates at weird angles,so it is basically unuseable

Rollback to the previous version everything is fine

I have compared the code , it seems that enabling Mahoney / Magiwick causing the problem btw

Nuzair46, 2023-01-18:
Had the same issue for me too.

unlogisch04, 2023-01-18:
I have the same observed with BMI160.

closed 2023-01-21 by Eirenliel

Nuzair46, 2023-02-14:
Can anyone confirm if v0.3.2 fixed this?

unlogisch04, 2023-02-14:
No it will be in v0.3.3 currently it is in main