SlimeVR/SlimeVR-Tracker-ESP #226: Make the logs have unique id for each message in front of each log line

loucass003, 2023-01-31:
In the gui or the server there is instances where we look for some specific lines of logs in the serial to do other actions.
it can causes issues if someone decides to change de text of that log line to something else and forget to change the server or gui that are lookig for it.

My proposal is that each log lines have their unique id. kinda like error id but for everything. Log lines that hold data would be the execption because you have to actually parse the log line to get data. but for all the others like

``looking for server...``
``handshake successfull``

it would be better to have something like

`` {1} looking for server ....``
`` {34} handshake successfull ...``

that way code inside the server could just use the id as a reference instead of the log text

Eirenliel, 2023-01-31:
Good suggestion, someone implement plz 🥺

Kamilake, 2023-02-01:
Does the firmware (not the server side) provide the id as above? If that's correct, server can easily figure out the stored Wi-Fi credentials or IMU state without parsing the serial console on the server side. Let's think about how the server can easily parse it without compromising readability.

unlogisch04, 2023-02-02:
The Firmware does not have any kind of text-IDs or event-IDs. Translations of the Firmware do currently break the detection in the WebFlasher. Probably also in the Server.
Current Log format is:
[LOGLEVEL] [Modul] Text\n
"[%-5s] [%s] %s\n" ( https://github.com/SlimeVR/SlimeVR-Tracker-ESP/blob/main/src/logging/Logger.cpp#L79 ) 

Maybe we can add a [Xabbcccc]
a = loglevel 4bit
b = modul 8bit
c = messageid 16bit (unique per modul?)

gorbit99, 2025-05-06:
@Eirenliel @unlogisch04 @loucass003 let's make the format of this concrete and then I'll probably implement it

gorbit99, 2025-05-08:
Please check #440 for a possible implementation