SlimeVR/SlimeVR-Tracker-ESP #227: A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header *** [upload] Error 2

backspacestud, 2023-02-04:
On one of my fully assembled trackers, when i upload the firmware it spits out this error
`Using manually specified: /dev/ttyS0
Uploading .pio/build/esp12e/firmware.bin
esptool.py v3.0
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
*** [upload] Error 2`

On a board without anything attached, it works.

For some reason the one with battery power shows up on the port "/dev/ttyS0" but on the one with nothing on it it shows up on "Serial port /dev/ttyUSB0". Could this be a power issue? should i upload the firmware before connecting the battery?

backspacestud, 2023-02-07:
Thank you for responding and sorry for the late response,

Board used: TP4056 USB C Charging board (https://www.aliexpress.us/item/2251832463465716.html?spm=a2g0o.order_list.order_list_main.35.21ef1802u5KKsC&gatewayAdapt=glo2usa&_randl_shipto=US)
MCU used: Wemos D1 Mini 4.0.0 (https://www.wemos.cc/en/latest/d1/d1_mini.html)
Define.h:
`#define IMU IMU_BMI160
#define SECOND_IMU IMU
#define BOARD BOARD_WEMOSD1MINI
#define IMU_ROTATION DEG_270
#define SECOND_IMU_ROTATION DEG_270

// Battery monitoring options (comment to disable):
// BAT_EXTERNAL for ADC pin, BAT_INTERNAL for internal - can detect only low battery, BAT_MCP3021 for external ADC
#define BATTERY_MONITOR BAT_EXTERNAL
#define BATTERY_SHIELD_RESISTANCE 180 //130k BatteryShield, 180k SlimeVR or fill in external resistor value in kOhm

#define PIN_IMU_SDA D2
#define PIN_IMU_SCL D1
#define PIN_IMU_INT D5
#define PIN_IMU_INT_2 D6
#define PIN_BATTERY_LEVEL A0`

I tried to follow the guide but as i am new to MCUs/Soldering in general I may have messed up somewhere. This is a diagram i tried to make because talking a photo of my rats nest of a tracker would have been too hard to understand. If there is anything I am leaving out I will do my best to provide more.
[image]

Thank you!

unlogisch04, 2023-02-07:
Make sure that there is no connection from pin 4 to pin 0. Pin 4 is pulled HIGH, but pin 0 need to be at GND for flashing (this happens over RTS/DTR lines on the D1 mini).

Also check 5 to RX. 

Im not sure why your tracker makes different types of USB devices.

backspacestud, 2023-02-18:
I have tried many different things and I don't think its the wiring, i think it has something to do with my linux install? not exactly sure so I tried to install it on windows and now im getting a new error, and since i have no background in VSCode I have no idea what it is asking for or how to fix it.

Whenever I try to build it Spits out the error 
`Transition by adding must_exist=False to SConscript calls.
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Missing SConscript 'C:\Users\USER\.platformio\packages\framework-arduinoespressif8266\tools\platformio-build.py'
LDF Modes: Finder ~ chain, Compatibility ~ soft
File "C:\Users\USER\.platformio\platforms\espressif8266\builder\frameworks\arduino.py", line 31, in <module>`

I tried looking it up and could not understand anything... Any help would be appreciated as I am running out of ideas.

backspacestud, 2023-02-18:
[{
	"resource": "/C:/Users/USER/.platformio/platforms/espressif8266/builder/frameworks/arduino.py",
	"owner": "_generated_diagnostic_collection_name_#2",
	"code": {
		"value": "reportMissingImports",
		"target": {
			"$mid": 1,
			"external": "https://github.com/microsoft/pyright/blob/main/docs/configuration.md#reportMissingImports",
			"path": "/microsoft/pyright/blob/main/docs/configuration.md",
			"scheme": "https",
			"authority": "github.com",
			"fragment": "reportMissingImports"
		}
	},
	"severity": 4,
	"message": "Import \"SCons.Script\" could not be resolved",
	"source": "Pylance",
	"startLineNumber": 27,
	"startColumn": 6,
	"endLineNumber": 27,
	"endColumn": 18
}]