SlimeVR/SlimeVR-Tracker-ESP #234: BMI160 Library dependency was not documented

RexaOnGit, 2023-03-20:
I discovered that the BMI160 library was copied and modified from the [following repository](
https://github.com/sgbihu/corelibs-arduino101/tree/master/libraries/CurieIMU/src). All credit for the register mapping, function names, and general structure goes to contributors on corelibs-arduino101 (which was forked by [sgbihu](https://github.com/sgbihu) from [arduino/ArduinoCore-arc32](https://github.com/arduino/ArduinoCore-arc32)).

There was no dependency added to the firmware, and no fork was created from the original repo. I have performed a diff between the original library and the one in SlimeVR-Tracker-ESP -- the changes are significant, and much of the code was deleted. It appears that the main point of the changes was to offload more of the code base to the I2C library made by Jeff Rowberg (which is [here](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/I2Cdev)). The screenshot below summarizes what happened fairly well:
[image]
Here is an example of how the remaining functions were changed:
[image]

Besides this, some blank lines and functions not used by SlimeVR were removed.

I'm not sure if any action is needed or can be taken, but I wanted to somehow document what happened and give credit to the original developer of the BMI160 library we use.

RexaOnGit, 2023-03-21:
UPDATE: The BMI160 library that was referenced is licensed by Intel under a General Public License (version 2.1). It is stated that we do not want to use any libraries with this type of license. What action is needed?

TheButlah, 2023-03-25:
re discord: Rexa is working on refactoring the bmi code to be based on an MIT licensed driver.

RexaOnGit, 2023-03-27:
My fork of SlimeVR firmware is [here](https://github.com/RexaOnGit/SlimeVR-Tracker-ESP/tree/main), and my fork of the new BMI160 library is [here](https://github.com/RexaOnGit/DFRobot_BMI160_SlimeVR/tree/master). I am about midway through refactoring with the new library.

RexaOnGit, 2023-03-31:
UPDATE: The DF robot library for BMI160 had too many issues. The new plan is as follows:

1. Make a general IMU communication manager (it will only pull in I2C dev for now)
2. Build a new BMI160 library from scratch
3. Validate the new library and communication manager
4. Split into dependencies

Progress will be visible in my [fork](https://github.com/RexaOnGit/SlimeVR-Tracker-ESP/tree/main).

closed 2025-05-14 by Eirenliel