SlimeVR/SlimeVR-Tracker-ESP #240: [  1606][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263

huangjun198604, 2023-05-08:
I use esp32 devkit v1 and bno085.
I can connected to bno085 but it doesn't respond.
Seem's like the problem of requestFrom(): i2cRead returned Error 263?

[log]

[INFO ] [SlimeVR] SlimeVR v0.3.3 starting up...
[INFO ] [Configuration] Found up-to-date configuration v1
[DEBUG] [Configuration] Found calibration data directory
[INFO ] [Configuration] Loaded configuration
[  1607][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[INFO ] [BNO080Sensor:1] Connected to BNO085 on 0x4b. Info: SW Version Major: 0x03 SW Version Minor: 0x02 SW Part Number: 0x98a6b4 SW Build Number: 0x06 SW Version Patch: 0x0d
[DEBUG] [SensorManager] No secondary IMU connected
[INFO ] [WiFiHandler] Setting up WiFi
[INFO ] [WiFiHandler] Loaded credentials for SSID  and pass length 8
[DEBUG] [WiFiHandler] Status: 6
[NOTICE] OTA updates allowed
.[ERROR] [BNO080Sensor:1] Sensor 1 doesn't respond. Last reset reason:
[ERROR] [BNO080Sensor:1] Last error: 0, seq: 0, src: 0, err: 0, mod: 0, code: 0
.[ERROR] [BNO080Sensor:1] Sensor 1 doesn't respond. Last reset reason:
[ERROR] [BNO080Sensor:1] Last error: 0, seq: 0, src: 0, err: 0, mod: 0, code: 0

Eirenliel, 2023-05-08:
Do you have INT pin connected, @huangjun198604? Since it connect to BNO but not read data, the problem is likely the INT pin.

huangjun198604, 2023-05-09:
> Do you have INT pin connected, @huangjun198604? Since it connect to BNO but not read data, the problem is likely the INT pin.

I conected and it works，thanks a lot！
but [log] [ 1607][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263 still here.
I think maybe it case by esp32 6.10 framework.

ChaimEvans, 2024-09-09:
> I conected and it works，thanks a lot！ but [log] [ 1607][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263 still here. I think maybe it case by esp32 6.10 framework.

I have the same problem, it doesn't work. How did you solve it?

Mikusho, 2025-01-14:
> I have the same problem, it doesn't work. How did you solve it?

I have same problem, I reballing the chip and it works!
as the master said, it's the INT pin.