SlimeVR/SlimeVR-Tracker-ESP #251: Support for SPI with BNO085 instead of I2C?

ReSummit, 2023-06-05:
I wanted to ask but is there a possibility to communicate with the BNO085 using SPI down the line? There is a ESP32 board that I have that has a BNO085 but is not connected through I2C.

If it hasn't been worked on yet, I've made some adjustments to the code in a fork that enables SPI communications but not sure if I should make a pull request or not.

ReSummit, 2023-06-26:
> Did you try any latency or precision tests?

I'm looking at this datasheet: https://cdn.sparkfun.com/assets/2/b/9/0/6/DS-14686-BNO080.pdf

On latency, it seems that from the BNO08X data sheet that SPI has lower latency depending on the frequency you use. Sections 5.2 and 5.3 show that I2C has timings in the scale of us whereas SPI has timings in the scale of ns (with the exception of two sections of timings involving Ns[?] and us). In general, SPI will provide higher bandwidth and lower latency due to the separate transmit and receive lanes.

On precision, it will basically be dependent on the sensor so nothing changes unless you increase the frequency for those. I'm slightly surprised that the frequencies for this are mostly 10 but maybe this is for battery life? You can technically get fast data in the SlimeVR Server and is probably doing that anyway.

The biggest problem if you do crank the latency as fast as possible is whether the router can handle all the data transmitted. Crank out as many measurements ignoring the battery life detriments but then now the latency bottleneck is on the networking side of things. Allowing the option though for high end setups could open the doors for higher accuracy though.

Tldr:
SPI lowers latency but does not affect precision unless you use that increased bandwidth for precision. It's best to allow for people to use SPI rather than I2C; then the bottleneck for latency becomes networking.

Shade-emry, 2025-02-24:
How did the testing for this work out?

Eirenliel, 2025-02-24:
There is already a branch that works with SPI for sfusion sensors: https://github.com/SlimeVR/SlimeVR-Tracker-ESP/tree/spi-support2 I will later add support for BNO.

Shade-emry, 2025-02-25:
Is the board already routed for SPI? if not have there been plans to route one? Also, I love the inductors and holes for EMI and thermal dissipation, it has a great design aspect. I always wondered why DIY versus slime official was behaving differently in closed tests, when I busted out the O-scope and logic analyzer and opened the schematic I immediately saw via the schematics the heavenly inductors and saw the holes in the board :) . Love those, they take bad noise and just turn it into heat. Can't get better then that.

Eirenliel, 2025-02-25:
@Shade-emry Yes there is. It's not final yet, so it's not published, but new revision of Slimes will use SPI.

Also the holes are not intentional, they're for physical compatibility with how historically SlimeVR trackers were developed, lol.

closed 2025-02-25 by Eirenliel