SlimeVR/SlimeVR-Tracker-ESP #254: Two commands for enabling features

ImUrX, 2023-06-29:
First is getting tracker features, this command should provide features by following this new type:
```
u8;[u32 <comma separated>]
ex: 10;32,52
```
(this is following the ASCII standard ofc)
The features are done via bitfields so each bit of the `u32` numbers means something, for now we will be using little endian for processing the data but its gonna be shown in the actual network packet as big endian but the spec would specify what each bit is so we are gonna use indexing for defining fields:
0. ``RUNTIME_MAG_SUPPORT``
1. ``MAG_ENABLED``
2. ``CALIBRATION_SUPPORT``
3. ``BLINK_SUPPORT``

So first bit (depends of endianness) tells you if the tracker itself has support for runtime magnetometer management #230 and second bit lets you know if it's enabled or disabled.

Second command is tracker trigger, this let's you trigger commands based on the previous feature, it has two arguments which the first one is based on `u16` (which is shorter than 256*32) and second one should only be a boolean for setting stuff to one or zero but we could flex this dependent to the first argument (**needs to be discussed**).
The first argument should refer to one of the index of tracker features (**needs to be discussed**) and it triggers that feature so you do ``TRIGGER 3`` and then tracker should blink. But this doesn't mean every feature triggers stuff like ``TRIGGER 0`` should do nothing.
``TRIGGER 1 true`` should set MAG_ENABLED to true and vice versa. 

This is a very flexible idea and shouldnt be that complicated to implement

ImUrX, 2023-06-29:
I must specify this is not a Serial command, its 2 network packets qwq

ImUrX, 2023-06-29:
the command stuff was to simplify explaining things, im sorry

ImUrX, 2023-06-29:
updated how the array would actually work ;-;

TheDevMinerTV, 2023-06-30:
I've started working on this in [impl/feature-retrieval](https://github.com/SlimeVR/SlimeVR-Tracker-ESP/tree/impl/feature-retrieval)

Eirenliel, 2023-06-30:
> I've started working on this in [impl/feature-retrieval](https://github.com/SlimeVR/SlimeVR-Tracker-ESP/tree/impl/feature-retrieval).

istg someone already did it for some feature, but i can't find which PR it's in...

nekomona, 2023-06-30:
> istg someone already did it for some feature, but i can't find which PR it's in...

like this one? #241

TheDevMinerTV, 2023-06-30:
We can integrate that into the code, I guess.