SlimeVR/SlimeVR-Tracker-ESP #277: BMI160 tracker only working with extra invisible IMU

backspacestud, 2023-08-19:
Hello, 
     I have been having issues getting my 2 IMU tracker to work, they are identified by slimeVR but the IMU's say working:false in the serial console. This is weird because my other tracker I got working yesterday was perfectly fine, but I decided to see if it was the firmware by changing the firmware of the working one. I could not remove the ghost IMU or else the actual one would not work at all, and the only defines.h i could get working was this:
[image]
(while this one would not work:)
[image]

When I tried to flash it to my second tracker It did not work at all, with or without the second IMU connected. My first question is why my 1st one only works with that specific defines.h and the second one does not work with anything at all (I triple checked the wiring though it may be a faulty sensor.)

Sorry for the confusing question, any help would be appreciated

nekomona, 2023-08-20:
Check the AD0 connection on the sensors, they need to match the address in the descriptor. For BMI160, `PRIMARY_IMU_ADDRESS_ONE` correspond to AD0 shorted to ground, while `PRIMARY_IMU_ADDRESS_TWO` correspond to AD0 floating.

backspacestud, 2023-08-20:
could it be this? 
```
[ERROR] [SensorManager] Can't find I2C device on provided addresses, scanning for all I2C devices and returning
[ERR] I2C: No I2C devices found

[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.1.42, mac: 48:3F:DA:5F:B8:D4, status: 2, wifi state: 1
[INFO ] [SerialCommands] Sensor[0]: BMI160 (0.000 0.000 0.000 1.000) is working: false, had data: false
[INFO ] [SerialCommands] Sensor[1]: BMI160 (0.000 0.000 0.000 1.000) is working: false, had data: false
```
And im not sure what you mean by AD0 connection (sorry im still new to microcontrollers/sensors like this). I checked my wiring based off of this diagram:
[image]
and the wiring of the working one was technically incorrect (I did not ground SAO). and when I tried to replicate the same thing on my non working sensor nothing changed... i'm struggling to find a common point of failure and any info that I can provide thats actually useful. (when my tracker with an auxiliary sensor fully match the diagram the tracker showed up but the sensors did not track). I will try to ask the discord about these issues as they may be able to help me with the BMI160 issues I have been having but I will do that at a later date. I may update this thread if i find out anything but it might be a while.

Muhammad-Hazimi-Yusri, 2023-09-10:
hi, thanks for this. I can confirm I am having same issue with BMI160 and Wemos D1 Mini, having both diode protection and battery level check resistor ast A0 (possibly the culprit),

using the defines.h code with non existant aux imu finally makes my imu working instad of error, while using only 1 imu without ghost aux imu, cause only error as if the IMU is broken.

thankfully i searched for related issue on github after anothr 2 of my imu isnt working which is too suspect.

osman-rodriguez, 2023-11-19:
Hi! Could you solve it? I have exactly the same problem...

unlogisch04, 2023-11-19:
> And im not sure what you mean by AD0 connection (sorry im still new to microcontrollers/sensors like this).

By AD0/SA0 he means the second ground wire on the main BMI160. This is the I2C Address selector for the BMI160.

unlogisch04, 2023-11-19:
> using the defines.h code with non existant aux imu finally makes my imu working instad of error, while using only 1 imu without ghost aux imu, cause only error as if the IMU is broken.

What setup do you have? (1x D1 mini 2xBMI160) or (1x D1 mini 1xBMI160)?
If you have (1x D1 mini 1xBMI160) then i can guess that the SA0 is not connected to ground. This changed not too long ago in the schematic (about 2 months i think).

On the firmware prior to 0.3.3 the firmware was less strict, and did not generate a ghost tracker when there was only 1 imu connected and that one did not have the correct i2c address. In the newer firmware it will create a ghost tracker for main and then assume that the other tracker is a extension.

If you have the sa0 grounded on the extension instead of the main, you need to change the following line in the defines.h:
from:
```cpp
#ifndef IMU_DESC_LIST
#define IMU_DESC_LIST \
        IMU_DESC_ENTRY(IMU,        PRIMARY_IMU_ADDRESS_ONE,   IMU_ROTATION,        PIN_IMU_SCL, PIN_IMU_SDA, PRIMARY_IMU_OPTIONAL,   PIN_IMU_INT) \
        IMU_DESC_ENTRY(SECOND_IMU, SECONDARY_IMU_ADDRESS_TWO, SECOND_IMU_ROTATION, PIN_IMU_SCL, PIN_IMU_SDA, SECONDARY_IMU_OPTIONAL, PIN_IMU_INT_2)
#endif
```
to:
```cpp
#ifndef IMU_DESC_LIST
#define IMU_DESC_LIST \
        IMU_DESC_ENTRY(IMU,        PRIMARY_IMU_ADDRESS_TWO,   IMU_ROTATION,        PIN_IMU_SCL, PIN_IMU_SDA, PRIMARY_IMU_OPTIONAL,   PIN_IMU_INT) \
        IMU_DESC_ENTRY(SECOND_IMU, SECONDARY_IMU_ADDRESS_ONE, SECOND_IMU_ROTATION, PIN_IMU_SCL, PIN_IMU_SDA, SECONDARY_IMU_OPTIONAL, PIN_IMU_INT_2)
#endif
```

hauveli, 2024-01-10:
Hello! I am writing because I had the exact same issue:
It connects to SlimeVR but shows "Error" with no description of why and no debug log (when connecting wirelessly).
Shorting SA0 to ground resolved the issue and I can now see it tracking.

I have also connected a BMM150 auxiliary magnetometer to it, but it is not picked up, should I open an issue for this?
Thank you for the messages above and for your time.

ZRock35, 2024-01-10:
BMM-150s don't work properly with BMI-160s currently, someone needs to figure that out.