SlimeVR/SlimeVR-Tracker-ESP #278: Tracker don't receive data [Wemos D1 Mini + MPU6050]

NintyS, 2023-08-24:
Hello I'm creating a set of trackers form Wemos D1 Mini and MPU6050 and IDK why I have something like that:
```
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
```
IDK why reacker don't receive any data.

I test it with i2c_scanner and MPU was found. GY-521 examples all works and return the data.

The defines.h file content:
```
#define IMU IMU_MPU6050
#define SECOND_IMU IMU
#define BOARD BOARD_WEMOSD1MINI
#define IMU_ROTATION DEG_180
#define SECOND_IMU_ROTATION DEG_90

#define MAX_IMU_COUNT 1

#ifndef IMU_DESC_LIST
#define IMU_DESC_LIST \
    IMU_DESC_ENTRY(IMU,        PRIMARY_IMU_ADDRESS_ONE,   IMU_ROTATION,        PIN_IMU_SCL, PIN_IMU_SDA, PIN_IMU_INT  ) 

#endif

// Battery monitoring options (comment to disable):
// BAT_EXTERNAL for ADC pin, BAT_INTERNAL for internal - can detect only low battery, BAT_MCP3021 for external ADC
#define BATTERY_MONITOR BAT_EXTERNAL
#define BATTERY_SHIELD_RESISTANCE 180 //130k BatteryShield, 180k SlimeVR or fill in external resistor value in kOhm

#define PIN_IMU_SDA 0
#define PIN_IMU_SCL 2
#define PIN_IMU_INT 14
#define PIN_IMU_INT_2 12
#define PIN_BATTERY_LEVEL A0
```
Is like that because with D's trackes don't achieve even second state of configuration so i use GPIOs.

Whole logs:
```
[INFO ] [SlimeVR] SlimeVR v0.3.3 starting up...
[INFO ] [Configuration] Found up-to-date configuration v1
[INFO ] [Configuration] Loaded configuration
[INFO ] [MPU6050Sensor:0] Connected to MPU6050 (reported device ID 0x68) at address 0x68
[DEBUG] [MPU6050Sensor:0] Enabling DMP...
[DEBUG] [MPU6050Sensor:0] DMP ready! Waiting for first interrupt...
[INFO ] [SensorManager] Sensor 1 configured
[INFO ] [SensorManager] 1 sensor(s) configured
[INFO ] [WiFiHandler] Setting up WiFi
[INFO ] [WiFiHandler] Loaded credentials for SSID dlink-BEB1 and pass length 11
[DEBUG] [WiFiHandler] Status: 7
[NOTICE] OTA updates allowed
...........[ERROR] [WiFiHandler] Can't connect from saved credentials, status: 7.
[DEBUG] [WiFiHandler] Trying saved credentials with PHY Mode G...
...........[ERROR] [WiFiHandler] Can't connect from saved credentials, status: 7.
[DEBUG] [WiFiHandler] Trying hardcoded credentials...
....[INFO ] [WiFiHandler] Connected successfully to SSID 'dlink-BEB1', ip address 192.168.100.145
[INFO ] [UDPConnection] Searching for the server on the local network...
[DEBUG] [UDPConnection] Handshake successful, server is 192.168.100.198:6969
[DEBUG] [UDPConnection] Server supports packet bundling
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
[NOTICE] OTA updates disabled by timeout, this is not an error
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
[INFO ] [SerialCommands] SlimeVR Tracker, board: 6, hardware: 1, build: 16, firmware: 0.3.3, address: 192.168.100.145, mac: A0:20:A6:23:15:21, status: 0, wifi state: 3
[INFO ] [SerialCommands] Sensor[0]: MPU6050 (0.000 0.000 0.000 0.000) is working: true, had data: false
```

Anything else what I can give?

unlogisch04, 2023-09-21:
Thanks for your report.
It is not working as you did mention. 
Please use the Firmware Version 0.3.2
currently in main it seems to be broken.

unlogisch04, 2023-09-22:
Ok, I got confused. 
For me the MPU6050 and WemosD1 mini works.
It just got down to 2 tps because of movement recolonization.

So you have probably a broken MPU or a new "version" or just a fake chip, that is not compatible.