SlimeVR/SlimeVR-Tracker-ESP #299: Firmware issues 0.3.3 and Main

SpudTree, 2023-10-08:
I tried both the Main and 0.3.3 firmware using the online firmware flasher but it didn't work with 2xbmi160 scenario. I downgraded to 0.3.2 and it worked.

When I used the standalone tracker it tracked, but when I attached the aux tracker only the aux tracked.

[INFO ] [SlimeVR] SlimeVR v0.3.3 starting up...
[INFO ] [Configuration] Found up-to-date configuration v1
[INFO ] [Configuration] Loaded configuration
[INFO ] [BMI160Sensor:0] Connected to BMI160 (reported device ID 0xd1) at address 0x68
[WARN ] [BMI160Sensor:0] No calibration data found for sensor 0, ignoring...
[INFO ] [BMI160Sensor:0] Calibration is advised
[WARN ] [GyroTemperatureCalibrator:0] No temperature calibration data found for sensor 0, ignoring...
[INFO ] [GyroTemperatureCalibrator:0] Temperature calibration is a work-in-progress feature; any changes to its parameters or updates will render the saved temperature curve invalid and unloadable.
[INFO ] [BMI160Sensor:0] Calibration data for gyro: not found
[INFO ] [BMI160Sensor:0] Calibration data for accel: not found
[INFO ] [BMI160Sensor:0] Initialized
[INFO ] [SensorManager] Sensor 1 configured
[INFO ] [BMI160Sensor:1] Connected to BMI160 (reported device ID 0xd1) at address 0x69
[WARN ] [BMI160Sensor:1] No calibration data found for sensor 1, ignoring...
[INFO ] [BMI160Sensor:1] Calibration is advised
[WARN ] [GyroTemperatureCalibrator:1] No temperature calibration data found for sensor 1, ignoring...
[INFO ] [GyroTemperatureCalibrator:1] Temperature calibration is a work-in-progress feature; any changes to its parameters or updates will render the saved temperature curve invalid and unloadable.
[INFO ] [BMI160Sensor:1] Calibration data for gyro: not found
[INFO ] [BMI160Sensor:1] Calibration data for accel: not found
[INFO ] [BMI160Sensor:1] Initialized
[INFO ] [SensorManager] Sensor 2 configured
[INFO ] [SensorManager] 2 sensor(s) configured
[INFO ] [WiFiHandler] Setting up WiFi
[INFO ] [WiFiHandler] Loaded credentials for SSID The Promised LAN 2.4 and pass length 17
[DEBUG] [WiFiHandler] Status: 7
[NOTICE] OTA updates allowed
...[INFO ] [WiFiHandler] Connected successfully to SSID
[DEBUG] [UDPConnection] Server supports packet bundling
[INFO ] [SerialCommands] Sensor[0]: BMI160 (0.000 0.000 0.000 1.000) is working: true, had data: false
[INFO ] [SerialCommands] Sensor[1]: BMI160 (0.011 -0.032 -0.697 -0.716) is working: true, had data: false
[NOTICE] OTA updates disabled by timeout, this is not an error
[INFO ] [SerialCommands] Sensor[0]: BMI160 (0.000 0.000 0.000 1.000) is working: true, had data: false
[INFO ] [SerialCommands] Sensor[1]: BMI160 (0.011 -0.032 -0.697 -0.716) is working: true, had data: false
[INFO ] [SerialCommands] Sensor[0]: BMI160 (0.000 0.000 0.000 1.000) is working: true, had data: false
[INFO ] [SerialCommands] Sensor[1]: BMI160 (0.011 -0.032 -0.697 -0.717) is working: true, had data: false

unlogisch04, 2023-11-18:
Thank you for your Report. Should be fixed in 0.4.0. Can you confirm that?

SpudTree, 2023-11-22:
Just tried 0.4.0, issue still remains