SlimeVR/SlimeVR-Tracker-ESP #300: Using together BMI160 + HMC5883L

Tanamichi, 2023-10-31:
Hello there!
Please tell me, it is possible use BMI160 + HMC5883L together? BMI160 it is great IMU for cheaper SlimeVR tracking, but all 6DOF IMU's affected to the drift.
It is possible use external magnetometer for drift compensation on BMI160? I saw on official website it working for MPU series IMU.
But on this repository in [README.md](https://github.com/SlimeVR/SlimeVR-Tracker-ESP#compatibility) we can see BMI160 compability section. Аlthough there is mention of a magnetometer. And we don't see any information about how to use it with.

How to do?
How to get working BMI160 + HMC5883L together?

Tanamichi, 2023-11-08:
any ideas on latest 0.4.0 firmware?

Tanamichi, 2023-11-19:
Found on Discord group:
Solder HMC SDA to BMI160 SDX, and solder HMC SCL to BMI160 SDX. Do not forget solder VCC and GND.
Check chip orientation via datasheet(do not use silkscreen printing on HMC pcb, it maybe can incorrect.
For woking HMC with BMI160 you need download firmware source and build manually(do not use online firmware builders) you need prepare firmware, here is 2 links from official wiki can help you.
[Setting up the Environment](https://docs.slimevr.dev/firmware/setup-and-install.html#setting-up-the-environment)
[Configuring the Firmware Project](https://docs.slimevr.dev/firmware/configuring-project.html#configuring-the-firmware-project)
After setting building environment and configure firmware you need to go next steps:
1. Open  `debug.h` and find 
```cpp
#define USE_6_AXIS true
```
change to 
```cpp
#define USE_6_AXIS false
```

2. Next step you need to go `defines_bmi160.h` and find this line. Uncomment this :)
```cpp// Pick one:
#define BMI160_MAG_TYPE BMI160_MAG_TYPE_HMC
```
These 2 lines enables magnetometer support for HMC5883L and BMI160
3. Build & Flash firmware.
4. After flashing do calibration.

if you feel with wrong magnetometer axes you can remap them:
Open `defines.h` and find Axis remapping example. 
Place this code after example:
```cpp
  #define BMI160_AXIS_REMAP_PRIMARY AXIS_REMAP_BUILD( \
  AXIS_REMAP_USE_X, \
  AXIS_REMAP_USE_Y, \
  AXIS_REMAP_USE_Z, \
  \
  AXIS_REMAP_USE_XN, \ 
  AXIS_REMAP_USE_YN, \
  AXIS_REMAP_USE_Z \
)
```

where `AXIS_REMAP_USE_XN` used for negation of X axis as example.

If you use default calibration methods(gyro calibrate 5 secs, 6 sided accelerometer calibration) you will have a more one step for calibrate your HMC. This step includes describing the infinity sign in the air for 20 seconds (with a tracker in hand)

if you finished calibration, tracker will be use magnetometer for drift correction.
I will close this issue.

closed 2023-11-19 by Tanamichi

C9Glax, 2024-01-18:
@Tanamichi Hey there, has this significantly improved tracking? Currently as soon as you leave the calibrated position, the drift gets very bad on the chnaged axis...

Tanamichi, 2024-01-19:
@C9Glax, Yeah, because magnetic environment is poor at your place. I have same issue, but if i'm stay in place were i'm calibrated i have stable results..
Also i have return to origin rotation were i pressed "Full Reset" in slime server.
For now best way use trackers without HMC or QMC with BMI160.
Cheers.

C9Glax, 2024-01-19:
Just to clarify:
It's better to **not** use HMC/QMC?
Thanks for the answer :)

Tanamichi, 2024-01-19:
@C9Glax Yes. For now do not use it :)

C9Glax, 2024-01-19:
Okay thank you! ❤️