SlimeVR/SlimeVR-Tracker-ESP #303: why after Working=false,there is no any sensor data

Kingwulin, 2023-11-13:
What is the purpose of setting working=false [here](https://github.com/SlimeVR/SlimeVR-Tracker-ESP/blob/main/src/sensors/bno080sensor.cpp#L180)? After setting false at this position in the code, it seems that there is no other place to set true.Working=false ESP32 has read sensor data, but it will not be read afterwards. Is this normal?

unlogisch04, 2023-11-13:
Hi
Yes, that is intended. Because if the sensor times out, there is something wrong that mostly can maybe solved with a reboot or not even with that (extension cable disconnected,..)
The BNO085 uses Interrupt Pins. If you have problems with the Sensor it might be, that the interrupt pin is assigned wrong. (From Version 0.4.0 the Interrupt pin gets handled a bit different when you have only 1 imu connected, and it does not have the correct address for the first one.

So more information would probably help.

Kingwulin, 2023-11-14:
> Hi Yes, that is intended. Because if the sensor times out, there is something wrong that mostly can maybe solved with a reboot or not even with that (extension cable disconnected,..) The BNO085 uses [...]

Yes, only one bno085 sensor is connected to ESP32, but I have also made the following settings

```
#define MAX_IMU_COUNT 1

// Axis mapping example
/*
#include "sensors/axisremap.h"
#define BMI160_QMC_REMAP AXIS_REMAP_BUILD(AXIS_REMAP_USE_Y, AXIS_REMAP_USE_XN, AXIS_REMAP_USE_Z, \
                                       AXIS_REMAP_USE_YN, AXIS_REMAP_USE_X, AXIS_REMAP_USE_Z)

IMU_DESC_ENTRY(IMU_BMP160, PRIMARY_IMU_ADDRESS_ONE, IMU_ROTATION, PIN_IMU_SCL, PIN_IMU_SDA, BMI160_QMC_REMAP) \
*/

#ifndef IMU_DESC_LIST
#define IMU_DESC_LIST \
        IMU_DESC_ENTRY(IMU,        SECONDARY_IMU_ADDRESS_TWO,   IMU_ROTATION,        PIN_IMU_SCL, PIN_IMU_SDA, PRIMARY_IMU_OPTIONAL,   PIN_IMU_INT)
        
#endif
```

unlogisch04, 2023-11-18:
If this is the full defines.h, the it is not complete. 

What i think could be the problem is `SECONDARY_IMU_ADDRESS_TWO`
As you did probably remove the line for `SECOND_IMU`

If you want the `PRIMARY_IMU_ADDRESS_ONE` or `PRIMARY_IMU_ADDRESS_TWO`
or directly `0x4A` or `0x4B`

Please make sure to follow the Schematic:
<https://docs.slimevr.dev/diy/tracker-schematics.html>
For the BNO itself you can look at that schematic: <https://oshwlab.com/slimevr/imu-breakout-board-bno085-rev-2-2>

suikabreaker, 2023-12-01:
I have the same issue. 1 BNO085 set to primary and connect with D1,D2,D5, address 0x4B. So I need to mannually assign the address?

unlogisch04, 2023-12-01:
Yes, if you don't want to solder you have to change the address. The online flasher does currently not support that.