SlimeVR/SlimeVR-Tracker-ESP #304: package.json  missing

jacob10001, 2023-11-18:
i keep getting this error message.
"MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package"

TheDevMinerTV, 2023-11-18:
What program is reporting this error?

jacob10001, 2023-11-18:
I'm using platformio in vs code

unlogisch04, 2023-11-18:
I don't think that it is a problem with the project. As a Project it is not common to need a "package.json".

Can you give more detailed information?
 - Did you restart your computer after installing VS Code, GIT, Platform IO?
 - Did you give PlatformIO enough time to download all its needs? 
 - How did you download the Project?
 - What commands did you run that the error did come up?

jacob10001, 2023-11-18:
i did not restart my computer after installing everything.
i tried two ways. downloading the .zip manually, and using the link to github.
after ensuring all necessary files were correct using the guide
it was when i clicked the build button that the error message appeared

unlogisch04, 2023-12-26:
Closing this issue. Reopen it if it is still a problem.

closed 2023-12-26 by unlogisch04