SlimeVR/SlimeVR-Tracker-ESP #335: Differentiate WiFi not found & failed to connect

ButterscotchV, 2024-06-25:
Since we can do a WiFi scan, surely we can report if the SSID was not found. Would make for more verbose logging and help with tech support and the GUI.

unlogisch04, 2024-06-25:
WiFi not found would work only when the SSID is not hidden. If the SSID is hidden, then we can not see if it does not exist or just not able to connect.
Any idea how you would want it?

ButterscotchV, 2024-06-26:
> WiFi not found would work only when the SSID is not hidden. If the SSID is hidden, then we can not see if it does not exist or just not able to connect.
> Any idea how you would want it?

I figure there's probably an error code somewhere? If not then that's dumb, idk what exactly to do in this case but my thought was just to have it print it out for now, so it wouldn't interfere if it does connect in the end

ButterscotchV, 2024-10-30:
We could also just mention it couldn't be found *or* the SSID is hidden. I imagine hidden SSID users are such a tiny percentage anyways and should know what they're doing.