SlimeVR/SlimeVR-Tracker-ESP #336: [Feature request] show tracker information in serial console log

kruemmelbande, 2024-08-02:
I think it would be helpful for providing tech support to just print the basic configuration of the firmware at startup, so for example the board, the imus, the pins etc. This would make identifying incorrectly flashed firmware much simpler.

ZycaR, 2024-08-15:
The firmwarte is capable of recognize `GET INFO` and `GET CONFIG` serial commands:
https://github.com/SlimeVR/SlimeVR-Tracker-ESP/blob/main/src/serial/serialcommands.cpp#L328
https://github.com/SlimeVR/SlimeVR-Tracker-ESP/blob/main/src/serial/serialcommands.cpp#L154-L161

Isn't this more request for GUI to add **Get Config** button to sertial console commands?
[image]

kruemmelbande, 2024-08-15:
Ehh, i mean, kinda, but i also just think its convenient if somebody asks for help and shows the serial console log that i dont have to specifically ask for get info or stuff like that. I get how its not *that* hard to ask for get info though, so i wouldnt exactly call this high priority