SlimeVR/SlimeVR-Tracker-ESP #370: The calculation of the system noise variance of the gyroscope bias should consider the update frequency of the gyroscope data, right?

Free-woodpecker, 2024-12-06:
[image]

Free-woodpecker, 2024-12-06:
My English is not very good. The above title was translated by AI.

Shade-emry, 2025-03-24:
This may not be an answer, but generally speaking when handling stuff like this i include Δt
Δt in the noise variance calculation for gyroscope bias to ensure proper scaling with update frequency. This is critical for accurate sensor fusion in Extended Kalman filters or similar algorithms.

Eirenliel, 2025-03-24:
@gorbit99 @kounocom probably @Free-woodpecker is right and line 902 need to replace `accTs` with `gyrTs`. It doesn't affect when it's the same Ts, but we're using different.

kounocom, 2025-03-24:
Already on it, will test soon

Eirenliel, 2025-03-30:
As explained by the author in https://github.com/dlaidig/vqf/issues/28#issuecomment-2764507908, this is intended. We also tried changing it to gyrTs and it gave worse result even after additional tuning.

closed 2025-03-30 by Eirenliel