SlimeVR/SlimeVR-Tracker-ESP #379: ICM45686 is not working properly on my new design

YuS1aN, 2025-01-27:
I have a slimevr board design and I have tried ICM20948 BMI270 on it. But when I replace the sensor to ICM45686, It doesn't work. 

Its logs seems like everything is fine... However, all values will not be updated.

[image]
[image]

The circuit is designed according to the datasheet.
[image]

Could someone tell me why?

gorbit99, 2025-01-27:
Current ICM45686 firmware assumes that the chip has a CLK attached to it, otherwise it won't work.

YuS1aN, 2025-01-28:
> Current ICM45686 firmware assumes that the chip has a CLK attached to it, otherwise it won't work.

Thanks for your answer. I am wondering how I can change it to use an internal CLK. I didn't find information about the register at 0x31 address in datasheet.

kounocom, 2025-01-28:
Just remove writes to 0x31 and 0x26 altogether

YuS1aN, 2025-01-28:
> Just remove writes to 0x31 and 0x26 altogether

It worked! Thanks a lot!

closed 2025-01-28 by YuS1aN