SlimeVR/SlimeVR-Tracker-ESP #387: [Feature Request] Ability to Offset the Rotation of Individual Trackers Using a Controller

Endakid, 2025-02-11:
I’ve noticed that drift usually affects only one SlimeVR tracker at a time. Since controllers have true 6DOF tracking and don’t experience this issue, it would be useful if we could virtually select/grab an individual tracker and manually adjust its rotation using a controller.

This would allow for quick corrections without needing a full reset, and it would work even while lying down. A standard full T-pose reset could clear any applied rotation offsets.

To keep adjustments precise and prevent overcorrection, it might be worth implementing a scaled-down effect—perhaps requiring two full rotations of the controller to rotate the tracker once. In my experience, only minor rotation offsets are needed to make the avatar look correct again.

Endakid, 2025-02-11:
> You could just use a yaw reset. That works while laying down too

True, but I find it sometimes makes it worse and doesn't fix the slight offset

unlogisch04, 2025-02-11:
I'm moving this issue to the server, as this tracker "offset" is handled there.