SlimeVR/SlimeVR-Tracker-ESP #414: ICM45686 I2C overtime

wsdmc, 2025-04-03:
When I use D1MINI and ICM45686 build a slime,The following message appears on the serial port
[INFO ] [SlimeVR] SlimeVR vgit-b873ba66dd39172f36ccdc1821151d6fc85a8814 starting up...
[INFO ] [Configuration] Found up-to-date configuration v1
[INFO ] [Configuration] Loaded configuration
[WARN ] [ICM-45686:0] No calibration data found for sensor 0, ignoring...
[INFO ] [ICM-45686:0] Calibration is advised
[INFO ] [ICM-45686:0] Sensor timesteps not calibrated
[INFO ] [ICM-45686:0] Gyro bias not calibrated
[INFO ] [ICM-45686:0] Accel bias not calibrated
[INFO ] [SensorManager] Sensor 1 configured
[INFO ] [SensorManager] 1 sensor(s) configured
[INFO ] [WiFiHandler] Setting up WiFi
[INFO ] [WiFiHandler] Loaded credentials for SSID 'SlimeVR' and pass length 10
[DEBUG] [WiFiHandler] Status: 7
[NOTICE] OTA updates allowed
.[ERROR] [ICM-45686:0] Sensor timeout I2C Address 0x68 delaytime: 2000 ms、
I connectedVIN GND SCL and SDA，does it need to connect the INT1 INT2 pin?

ShineBrightMeow, 2025-04-03:
Please refer to https://docs.slimevr.dev/diy/tracker-schematics.html

VIN is not used, but 3V3 is. Also, make sure the SD0 pads are bridged together.

For any other DIY assembly issues, please ask in `#diy` channel on SlimeVR Discord (https://discord.gg/SlimeVR)

wsdmc, 2025-04-04:
I made a mistake, LWKJ's ICM45686,only has VCC, no VIN and 3V3,SD0 already connected,

wsdmc, 2025-04-05:
I found YBX 45686 have the same problem

ShineBrightMeow, 2025-04-05:
Try `kounocom/dynamic-sfusion-icm45-hotfix` firmware from my webflasher. (https://slimevr.shinebright.dev/)

unlogisch04, 2025-04-05:
Make sure you wire the modules correctly
For I2C you have to pull up the CS pin (pad 12) to vddio
And also select the correct address with (pad 1) AP_SDO AP_AD0
see the datasheet chapter 4.4:
<https://invensense.tdk.com/wp-content/uploads/documentation/DS-000577_ICM-45686.pdf>
[image]

For YBX-ICM45686:
VCC -> 3.3V
GND -> GND
SCK -> SCL
MOSI -> SDA
MISO -> ADO/SDO (address selector)
CSB -> VDDIO (or pull up to 3.3V should probably be ok)

wsdmc, 2025-04-13:
> Try firmware from my webflasher. (https://slimevr.shinebright.dev/`kounocom/dynamic-sfusion-icm45-hotfix`)

it seem that can't be calibrated,I try bmi160 and mpu 9250's way, but it not work
[INFO ] [SlimeVR] SlimeVR vgit-f49f3ba30d9828e1a82d225fd8a964d490c2de89 starting up...
[INFO ] [Configuration] Found up-to-date configuration v1
[DEBUG] [Configuration] Found sensor calibration for NonBlocking (common) at index 0
[INFO ] [Configuration] Loaded configuration
[INFO ] [SensorManager] Sensor 1 found at address 0x68
[INFO ] [DynamicCalibration] Calibrated timesteps: Accel inf, Gyro inf, Temperature 0.000309
[INFO ] [DynamicCalibration] Gyro bias not calibrated
[INFO ] [DynamicCalibration] Accel bias not calibrated
[INFO ] [SensorManager] Sensor 1 configured
[INFO ] [SensorManager] 1 sensor(s) configured
[INFO ] [WiFiHandler] Setting up WiFi
[INFO ] [WiFiHandler] Loaded credentials for SSID SlimeVR and pass length 10
[DEBUG] [WiFiHandler] Status: 7
[NOTICE] OTA updates allowed

wsdmc, 2025-04-13:
even i move tarcker，Acceleration and rotation   not change

wsdmc, 2025-04-13:
the model's seller told me cs alread in Highlevel

wsdmc, 2025-04-21:
I tried esp32c3 supermini，It has the same problem

[INFO ] [SlimeVR] SlimeVR vgit-200681091 starting up...
[INFO ] [Configuration] Found up-to-date configuration v1
[INFO ] [Configuration] Loaded configuration
[WARN ] [ICM-45686:0] No calibration data found for sensor 0, ignoring...
[INFO ] [ICM-45686:0] Calibration is advised
[INFO ] [ICM-45686:0] Gravity read: 0.0 (need < -7.5 to start calibration)
[INFO ] [SensorManager] Sensor 1 configured
[INFO ] [SensorManager] 1 sensor(s) configured
[INFO ] [WiFiHandler] Setting up WiFi
[INFO ] [WiFiHandler] Loaded credentials for SSID '' and pass length 10
[DEBUG] [WiFiHandler] Status: 6
[NOTICE] OTA updates allowed
.[ERROR] [ICM-45686:0] Sensor timeout I2C Address 0x68 delaytime: 2005 ms
.[INFO ] [WiFiHandler] Connected successfully to SSID 'SlimeVR', IP address 192.168.43.243
[INFO ] [UDPConnection] Searching for the server on the local network...
[DEBUG] [UDPConnection] Handshake successful, server is 192.168.43.1:6969
[DEBUG] [UDPConnection] Server supports packet bundling

gorbit99, 2025-04-23:
@wsdmc Can you link this module? It sounds a lot like it doesn't have a clock attached, which makes it incompatible with slimevr ESP driver currently.

wsdmc, 2025-05-15:
> [@wsdmc](https://github.com/wsdmc) Can you link this module? It sounds a lot like it doesn't have a clock attached, which makes it incompatible with slimevr ESP driver currently.

https://e.tb.cn/h.6riubgBrfBCWdzl?tk=MBsHVi69GKR HU926 
there it is,sorry for reply too late,I stayed at school for the past three weeks and couldn't bring my phone

gorbit99, 2025-05-15:
Indeed that is the issue, it doesn't have a clock (it would be a shiny rectangle otherwise), and that is currently not supported by the ESP firmware.

wsdmc, 2025-05-16:
> Indeed that is the issue, it doesn't have a clock (it would be a shiny rectangle otherwise), and that is currently not supported by the ESP firmware.

ok,thank for answer,I'll try another model

closed 2025-05-16 by wsdmc