SlimeVR/SlimeVR-Tracker-ESP #426: Machine optimized code

Shade-emry, 2025-04-23:
[image]

I thought AI or Sim-based code was not accepted here?

Shade-emry, 2025-04-23:
static constexpr VQFParams SensorVQFParams{
	.tauAcc = 0.01f, // ultra-responsive
	.biasSigmaInit = 1.0f,
	.biasForgettingTime = 50.0f,
	.biasClip = 8.0f,
	.biasSigmaMotion = 0.5f,
	.biasVerticalForgettingFactor = 0.1f,
	.biasSigmaRest = 0.1f,
	.restMinT = 0.1f,  // detects rest fast if ever used
	.restFilterTau = 0.05f,
	.restThGyr = 3.0f,
	.restThAcc = 1.5f,
};

this should work out the issues with the latency on movement supposedly according to matlab2024b

Eirenliel, 2025-04-23:
Maybe if you knew the difference between machine learning and generative AI. Or could read code and understand what has been done in that PR.

closed 2025-04-23 by Eirenliel

Shade-emry, 2025-04-23:
MatLab2024b is not generative ai, it has to have real world movement samples. It can run sims with various inputs based on that, granted the code I paid for was shit, it was not my own, so I would appreciate it if you took you attitude somewhere else as I'm LEARNING to code and don't think I'm this god coder.

sctanf, 2025-04-24:
Hi @Shade-emry, gen AI or sim was not used to gather parameters or generate data. The new parameters are optimized by a differential evolution algorithm, and the data used is recorded from physical devices.

Shade-emry, 2025-04-24:
Is this in the utilization of that neat 3-axis turntable I saw Eiren post?
Or just raw capture without any turntable from just the sensor?

Shade-emry, 2025-04-24:
This sounds ALOT like numpy. Thank you for the input.

sctanf, 2025-04-24:
Hello @Shade-emry, the raw sensor data is captured while the device is in normal use. In the future, turntable may be used.