SlimeVR/SlimeVR-Tracker-ESP #427: Trackers cannot connect to 802.11N-only network but can connect to G networks

i0ntempest, 2025-04-24:
I have a standalone small router acting as the AP for SlimeVR trackers only. It's the AR300M16 from GL.iNet, running a customized OpenWRT. When I set the network to N only, none of my trackers with 0.5.4 or 0.5.0 firmware will connect. I must set the network to b/g for the trackers to be able to connect.
[This](https://github.com/nodemcu/nodemcu-firmware/issues/3017) might be related, it's solved in a new SDK but my board (`BOARD_WEMOSD1MINI`) does not pull external SDK at all.
Help please?

Eirenliel, 2025-04-24:
If you're using it for SlimeVR specifically, just keep it on mixed, what's the problem?

We will look into this though, because it might affect users who use normal wifi setups.

i0ntempest, 2025-04-24:
> If you're using it for SlimeVR specifically, just keep it on mixed, what's the problem?
> 
> We will look into this though, because it might affect users who use normal wifi setups.

Well it's not a big problem, but I think N is a tiny bit better on latency and stuff, noticable or not.
Also I just like tinkering with networking stuff :)

0forks, 2025-04-24:
I'm running slimes on an N-only 20mhz network, channel 1 (MikroTik hAP ac²), everything connects fine.
Could you list all AP settings you have access to in OpenWRT so we can try to reproduce?

i0ntempest, 2025-04-25:
This is the output of `uci show wireless` when in N only mode:
```
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/ahb/18100000.wmac'
wireless.radio0.band='2g'
wireless.radio0.disabled='0'
wireless.radio0.hwmode='11g'
wireless.radio0.htmode='HT20'
wireless.radio0.channel='1'
wireless.radio0.cell_density='0'
wireless.radio0.country='AU'
wireless.radio0.distance='5'
wireless.radio0.beacon_int='25'
wireless.radio0.require_mode='n'
wireless.radio0.legacy_rates='0'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.encryption='psk2'
wireless.default_radio0.wds='1'
wireless.default_radio0.ifname='wlan0'
wireless.default_radio0.key='removed'
wireless.default_radio0.ssid='SlimeVR_I0N'
wireless.default_radio0.dtim_period='1'
wireless.default_radio0.hidden='0'
wireless.guest2g=wifi-iface
wireless.guest2g.device='radio0'
wireless.guest2g.network='guest'
wireless.guest2g.mode='ap'
wireless.guest2g.ifname='wlan0-1'
wireless.guest2g.encryption='psk2'
wireless.guest2g.key='removed'
wireless.guest2g.ssid='GL-AR300M-522-Guest'
wireless.guest2g.guest='1'
wireless.guest2g.disabled='1'
wireless.guest2g.wds='1'
wireless.guest2g.isolate='1'
```
If I switch to g/n mixed mode then `wireless.radio0.require_mode='n'` is removed and trackers can connect.

gorbit99, 2025-05-04:
I found an issue that's very much like this one while developing some stuff. Seems to be a problem with the ESP8266 specifically when paired with some routers. Specifically the mcu connects to the wifi just fine, but then fails when trying to get an ip from the network. I don't think this is solvable.