SlimeVR/SlimeVR-Tracker-ESP #449: ICM-42688-P data abnormal

c0n57, 2025-05-16:
Using SlimeVR-Server 0.14.1, Windows 10 Desktop
Using SlimeVR-Tracker-ESP 0.5.4
Using ESP-12F + ICM-42688-P(I428 for short), chip silk printing I428P B23AA1 2434 (year 2024, week 34)

I' m using my own designed PCB, and it works well with BMI270. But if I use I428 instead, the data just jumps irregularly and speedy. The rotation X/Z jumps between +-180, Y between +-100. **Also I find that the tempreture , Temp. C line in the server, is relatively close to 0 (or say, between +-10) when the tracker is static. If I shake the tracker rapidly, the rotation data jumps as usual, but the tempreture data reaches a huge number like -5400 or +2000.** If my PCB contains any problem please figure them out. Or, I think the data processing for I428 may contain error, furthermore the ICM45686 may contain the same problem.

[image]

c0n57, 2025-05-16:
Designator PWR and BAT in the schematic are for power switch and battery. In fact, ICM42688, ICM45686 and BMI270 share the same packaging type, they can be simply replaced by each other while soldering them(I think?).

Eirenliel, 2025-05-16:
First of all, you need to have a clock for ICM45686 to work. This can be disabled in firmware. Either way, for ICM you shouldn't use 0.5.4 firmware, you should either use main or some of the other branches that are in development right now. There haven't been an official release of firmware that supports ICM well.

unlogisch04, 2025-05-16:
Seems like Main is broken for ICM-42688-P
Test with version V0.5.4 -> works
Test with main -> broken

kounocom, 2025-05-16:
Hi-res mode for ICM-42688-P was attempted to be implemented with dynamic-sfusion, but we couldn't get it working and the whole PR got merged with those incomplete changes. It should be fixed properly, but a quick fix would just be reverting the driver to before dynamic-sfusion got merged

c0n57, 2025-05-16:
> Seems like Main is broken for ICM-42688-P Test with version V0.5.4 -> works Test with main -> broken

Oh you remind me that I'm actually using main but not release. I'll try release 0.5.4 later to see if it works.

_(also i spell the word temperature wrong omg what am i doing)_

c0n57, 2025-05-16:
> First of all, you need to have a clock for ICM45686 to work. This can be disabled in firmware. Either way, for ICM you shouldn't use 0.5.4 firmware, you should either use main or some of the other [...]

ummm I just find that I'm using main actually, I'll try release later. If release works then this doesn't seem to be a release issue.

Anyway, in TDK's datasheet, an internal clock is available. If I want to add a 32.768kHz oscillator, can I just connect the oscillator's output to INT2/CLKIN pin?

TDK's I428 datasheet DS00347, pg 14 and 24.
[image]

[image]

c0n57, 2025-05-16:
SCH looks like this, is it correct?

[image]

c0n57, 2025-05-17:
Update now, I use release 0.5.4 instead and it works well with I428 or I458. It seems that main branch is still developing for ICM chips. SCH is acceptable but I'll try add an oscillator anyway. Thx for everyone's reply :-)

closed 2025-05-17 by c0n57