SlimeVR/SlimeVR-Tracker-ESP #467: ICM45686 mag support problem

VegetablCat, 2025-08-13:
The mag-support branch used, IMU is ICM45686 and magnetometer is QMC6309. After connecting to the server, the magnetometer switch controlling a single tracker will not be displayed. When the tracker works normally for a period of time, it can't get data from IMU, and the serial port shows that all three axes are nan, so it can work normally only after reset. Moreover, we don't know what the trigger condition is, and it will be possible to identify it as a tracker supporting magnetometer after a period of time, but even if it is not identified as supporting magnetometer, the magnetometer data will still be valid.

gorbit99, 2025-08-15:
Mag support is not a feature ready for testing or daily use

closed 2025-08-15 by gorbit99

brightproject, 2025-12-23:
> The mag-support branch used, IMU is ICM45686 and magnetometer is QMC6309. 

https://github.com/tdk-invn-oss/motion.arduino.ICM45686/issues/7